prampec/arduino-softtimer #6: Form Arduino version 1.6.6 got error: 'callBack' was not declared in this scope

prampec, 2015-11-13:
Form Arduino version 1.6.6 all previous codes, including all examples fails with a compiler error message like this: 'callBack' was not declared in this scope

prampec, 2015-11-13:
This is because it seem that from Arduino 1.6.6 functions needs to be defined **before** using them as function pointers.
I still need to investigate whether this is intended to be a change from 1.6.6, of will be fixed in later versions.

For a quick fix you need to do define a function first like this:
```c
void callBack1(Task* me);
Task t1(2000, callBack1);
```

If this is a permanent change in Arduino, than all documentation and all Example code needs to be updated accordingly.

prampec, 2015-11-13:
This issue will not be fixed in SoftTimer, rather we will wait Arduino to fix this issue in the compiler in later versions.
See details: https://github.com/arduino/arduino-builder/issues/50

PhilRounds, 2016-02-05:
That seems to be it, thanks.  Of course now it can't find PciListener.h somewhere further down the line, but I'm guessing that's my issue.

container, 2016-02-09:
I have similar error, after moving the Task t1(0, myLoop) below the function definition, as you have suggested: 
"Arduino\libraries\SoftTimer\src\Debouncer.h:32:25: fatal error: PciListener.h: No such file or directory."

PhilRounds, 2016-02-09:
In Debouncer.h there are two lines:

//#include <PciListener.h>
#include "../PciManager/PciLIstener.h"

Switch which of these two is commented out, and PciListener.h gets included correctly.

ContainedShadow, 2017-01-12:
I tried switching the one that is commented. But I now I get this error: 

In file included from C:\Users\Name\Documents\Arduino\libraries\SoftTimer\src\Debouncer.cpp:28:0:

C:\Users\Name\Documents\Arduino\libraries\SoftTimer\src\Debouncer.h:33:39: fatal error: ../PciManager/PciListener.h: No such file or directory

 #include "../PciManager/PciListener.h"

prampec, 2017-06-11:
Finally I decided to fit all the examples of the SoftTimer to the current requirements of the Arduino compiler, as the authors does not seem to to solve this problem.

closed 2017-06-11 by prampec