prampec/arduino-softtimer #12: SoftTimer6Rotary example

pasicemir, 2016-05-27:
Would it be possible to publish SoftTimer6Rotary example with changes that makes compilations possible in latest Arduino version. I tried including function declarations "void callBack(Task* me);" but I could not make it compile. Compiler comes with error message:
exit status 1
invalid conversion from 'void (*)(Task*)' to 'void (*)(short int, Rotary*)' [-fpermissive]

prampec, 2016-05-27:
I'm sorry to hear that this bug is still not fixed in the current version of Arduino.
Until it is not fixed, you need to declare all functions before referencing them.
E.g. in this case you put this before creating Rotary instance "r":
```
void onRotate(short direction, Rotary* rotary);
void onRotPushPress();
void onRotPushRelease(unsigned long pressTime);
```
See issue #6 , also please add comment to this issue with your Arduino version.

closed 2016-05-27 by prampec

pasicemir, 2016-05-27:
Thank you very much, for such a fast response.