prampec/arduino-softtimer #13: Strange compile error with SoftTimer6Debounce1

GregWoods, 2016-06-20:
when using SoftTimer6Debounce1 either from the Examples folder or pasted from github, I get the following compile error (verbose) - shown at the end of this message.
However, if I move the onPressed() function to above the "Debouncer" line, it compiles fine.
Oddly, I do not get the same error on the onReleased() function.

Any ideas?

```
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Greg\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Greg\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Greg\Documents\Arduino\libraries" -fqbn=arduino:avr:leonardo -vid-pid=0X2341_0X8036 -ide-version=10609 -build-path "C:\Users\Greg\AppData\Local\Temp\buildaadc362bdf274423cf36f8a56937bf07.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\examples\SoftTimer6Debouncer1\SoftTimer6Debouncer1.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\Greg\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\Greg\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\Greg\Documents\Arduino\libraries" -fqbn=arduino:avr:leonardo -vid-pid=0X2341_0X8036 -ide-version=10609 -build-path "C:\Users\Greg\AppData\Local\Temp\buildaadc362bdf274423cf36f8a56937bf07.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\examples\SoftTimer6Debouncer1\SoftTimer6Debouncer1.ino"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "C:\Users\Greg\AppData\Local\Temp\buildaadc362bdf274423cf36f8a56937bf07.tmp\sketch\SoftTimer6Debouncer1.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src" "C:\Users\Greg\AppData\Local\Temp\buildaadc362bdf274423cf36f8a56937bf07.tmp\sketch\SoftTimer6Debouncer1.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\src" "C:\Users\Greg\AppData\Local\Temp\buildaadc362bdf274423cf36f8a56937bf07.tmp\sketch\SoftTimer6Debouncer1.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\src" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src" "C:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src\PciListenerImp.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\src" "C:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src\PciListenerImp2.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\src" "C:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src\PciManager.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\src" "C:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\src\BlinkTask.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\leonardo" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master\src" "-IC:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\src" "C:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master\src\Debouncer.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR  -DUSB_VID=0x2341 -DUSB_PID=0x8036 '-DUSB_MANUFACTURER="Unknown"' '-DUSB_PRODUCT="Arduino Leonardo"' "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants [... cut ...]
SoftTimer6Debouncer1:8: error: 'onReleased' was not declared in this scope

 Debouncer debouncer(INPUT_PIN, MODE_CLOSE_ON_PUSH, onPressed, onReleased);

                                                               ^

Using library arduino-pcimanager-master at version 2.1.0 in folder: C:\Users\Greg\Documents\Arduino\libraries\arduino-pcimanager-master 
Using library arduino-softtimer-master at version 3.1.0 in folder: C:\Users\Greg\Documents\Arduino\libraries\arduino-softtimer-master 
exit status 1
'onPressed' was not declared in this scope
```

GregWoods, 2016-06-20:
To answer my own question...
It seems to be an Arduino IDE problem.
[http://forum.arduino.cc/index.php?topic=42835.0](http://forum.arduino.cc/index.php?topic=42835.0)
Tried in 1.6.7 and 1.6.9, and both have the problem.

Adding prototype definitions for onPressed and onReleased before the Debouncer line also works 
i.e.
```
void onPressed();
void onReleased(unsigned long);
//the above lines are no substitute for the real onPressed and onReleased functions
Debouncer debouncer(INPUT_PIN, MODE_CLOSE_ON_PUSH, onPressed, onReleased);
```

prampec, 2016-12-06:
Right. Also mentioned in issue #6

closed 2016-12-06 by prampec