prampec/arduino-softtimer #14: Compatibility with Arduino Zero boards

trlafleur, 2016-07-18:
In testing with a Zero 32bit processor I get errors like this...
So it appears that this fine library is yet NOT compatible with this processor.

```
arduino-softtimer-master/src/BlinkTask.cpp:38:17: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   _portRegister =  portOutputRegister(digitalPinToPort(outPin));
```

trlafleur, 2016-07-18:
fyi...

I was able to make it run on a zero by changing all of the below statement to int32_t in a number of the .h files
I have only done a very limited test... this may not be the correct long term fix as its just for the Zero M0 processor

    //volatile uint8_t *_portRegister;
    volatile uint32_t *_portRegister;

And for now, I also needed to remove the function of Debouncer, and Rotary that call pciManager from the library

TheeCaptain, 2017-05-07:
I am using the Feather Huzzah (ESP8266 with some extras) and getting the same problem.

```
SoftTimer\src\BlinkTask.cpp:38:17: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

   _portRegister = portOutputRegister(digitalPinToPort(outPin));
```

prampec, 2017-05-07:
Fixed in version 3.1.1.
Please also update the PciManager library (2.1.1 or later) from the LibraryManager.
Please also note, that PCI related options are disabled on a hardware, where PCINT vect. not supported.

closed 2017-06-11 by prampec

jishindev, 2018-07-18:
Have this same issue with ESP32.
`BlinkTask.cpp: In member function 'void BlinkTask::init(byte, long unsigned int, long unsigned int, byte, long unsigned int)':
BlinkTask.cpp:42:23: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   this->_portRegister = portOutputRegister(digitalPinToPort(outPin));
                       ^`

Gixy31, 2020-12-06:
Any fix for ESP32? Same here with PCIManager 2.1.4 and SoftTimer 3.2.0

Gixy31, 2020-12-06:
Solved with this in some .h files 👍 
#if (!defined ESP8266 && !defined ESP32) 
    volatile uint8_t *_portRegister;
#else
    volatile uint32_t *_portRegister;
#endif