prampec/arduino-softtimer #15: "SoftTimer6Debouncer1.ino" doesn't work

XxPhenixX, 2016-11-03:
Hello,

the code doesn't react to a HIGH on Pin 3.
The "Ready." is shown in the Terminal, so the Communication should work.

What is the Problem?

Thanks!
Martin

prampec, 2016-11-03:
Hi Martin,
I need more details on this.
This example assumes, that you have an input (e.g. a switch) attached to the Arduino pin 3. (Like in this tutorial: https://www.arduino.cc/en/Tutorial/DigitalReadSerial)

XxPhenixX, 2016-11-03:
Yes exactly like that, a pushbutton.
I measured on the pin 3 => 5V.

XxPhenixX, 2016-11-03:
Btw. I have exactly the same problem with your example for the "arduino-pcimanager" :(

prampec, 2016-11-03:
Please be a bit more precise!
You start the SoftTimer6Debouncer1 example.
You have the GND connected to pin 3 via a pulldown resistor of 10KOhm.
Now you measure 0 volt on pin 3.
Now you also connect  VCC to pin 3 via a tactile switch. Now you measure 5 volts on pin 3. As this is pin change, you expect the example to inform you about that, but you only see the initialization "Ready" message.
Please confirm this!
Also: Please provide the exact Arduino hardware type and the IDE you are using.

XxPhenixX, 2016-11-04:
- Arduino IDE 1.6.12

- Arduino Board Mega2560 A000067

- Yes. First I measured 0V than I press the button, than 5V (with "digitalread" it works fine)

My circuit is like that:

[image]

prampec, 2016-11-04:
Would you please be so kind, and test the SoftTimer (and PciManager) with a regular push button setup instead a complex circuit.
Also note, that some examples might use the internal pullup resistor of the Arduino that might modify your circuit. (Debouncer also support pullups with a 5th argument in the constructor.)

XxPhenixX, 2016-11-06:
Okay, yes I could try that on Tuesday or Wednesday!

closed 2016-12-06 by prampec