prampec/arduino-softtimer #19: SoftTimer1 Task - crashes

Pfannex, 2017-06-24:
Hi,
I try to run this arduino sketch on my ESP8266, but the sketch crashes.....
Any ideas why?

```c++
#include <SoftTimer.h>

// -- Define method signature.
void callBack1(Task* me);
void callBack2(Task* me);

Task t1(1000, callBack1);
Task t2(5000, callBack2);

void setup() {
  Serial.begin(115200); 
  Serial.println("");
  SoftTimer.add(&t1);
  SoftTimer.add(&t2);
}

void callBack1(Task* me) {
  Serial.println("Task1");
}
void callBack2(Task* me) {
  Serial.println("Task2");
}
```

Serial log:
```
Task1
Task2
Task1
Task1
Task1

Soft WDT reset

ctx: cont 
sp: 3ffef1f0 end: 3ffef400 offset: 01b0

>>>stack>>>
3ffef3a0:  3ffe8468 0031d1a7 3ffee2cc 40201bef  
3ffef3b0:  3ffe8638 feefeffe 3ffee3b4 40201cf6  
3ffef3c0:  feefeffe 3ffee2e0 3ffee2b8 40201d59  
3ffef3d0:  3fffdad0 00000000 3ffee3d0 40201d6f  
3ffef3e0:  feefeffe feefeffe 3ffee3d0 40202148  
3ffef3f0:  feefeffe feefeffe 3ffee3e0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
⸮
```

prampec, 2017-06-24:
Very strange. Seems to be some kid of watchdog functionality.
Please try these methods: Disable watchdog, feed the watchdog,
as described here: https://techtutorialsx.com/2017/01/21/esp8266-watchdog-functions/

Pfannex, 2017-06-25:
OK, this seems to work....

```c++
void callBack1(Task* me) {
  ESP.wdtFeed();
  Serial.println("Task1");
}
void callBack2(Task* me) {
  ESP.wdtFeed();
  Serial.println("Task2");
}
```

prampec, 2017-06-26:
Thanks to highlight this behavior out!

closed 2017-06-26 by prampec