prampec/arduino-softtimer #21: ESP8266 NodeMcu Soft WDT reset error

tommyNy, 2017-08-20:
Code from examples, but with Serial.println

#include <SoftTimer.h>

// -- Define method signature.
void callBack1(Task* me);
void callBack2(Task* me);

Task t1(2000, callBack1);
Task t2(1111, callBack2);

void setup() {
  Serial.begin(9600);
  
  SoftTimer.add(&t1);
  SoftTimer.add(&t2);
}

void callBack1(Task* me) {
  Serial.println("callBack1");
}
void callBack2(Task* me) {
  Serial.println("callBack2");
}

Output:
callBack1
callBack2
callBack2
callBack1
callBack2

Soft WDT reset

ctx: cont 
sp: 3ffef1c0 end: 3ffef3b0 offset: 01b0

>>>stack>>>
3ffef370:  feefeffe 3ffee290 3ffee27c 40201d51  
3ffef380:  3fffdad0 00000000 3ffee380 40201d67  
3ffef390:  feefeffe feefeffe 3ffee380 40202140  
3ffef3a0:  feefeffe feefeffe 3ffee390 40100114  
<<<stack<<<
3⸮⸮⸮⸮

prampec, 2017-08-20:
Please visit page https://github.com/prampec/arduino-softtimer/blob/wiki/WhyHow.md#platform-dependent-notes for ESP8266 watchdog workaround.

closed 2017-08-20 by prampec

prampec, 2017-08-20:
I also have added an ESP8266 example for you: https://github.com/prampec/arduino-softtimer/blob/master/examples/SoftTimer1TaskESP8266/SoftTimer1TaskESP8266.ino

tommyNy, 2017-08-20:
Ok, thanks. I tested it with disabled watchdog, but better it works with feed function.