prampec/arduino-softtimer #22: ESP8266 NodeMcu doesn't work with WiFi

tommyNy, 2017-08-27:
I can't use wifi parallel with a soft timer lib. I can see console output with wifi status OK, but I can't ping ESP module. When I remove a soft timer lib from code, WiFi connection work fine, I can ping module.

`#include <SoftTimer.h>
#include <ESP8266WiFi.h>

// -- Define method signature.
void feedWatchdog(Task* me);
void callBack1(Task* me);
void callBack2(Task* me);

Task watchdogFeederTask(200, feedWatchdog);
Task t1(3000, callBack1);
Task t2(5000, callBack2);

char ssid[] = "ssid";
char wifiPass[] = "pass";

IPAddress ip(192, 168, 0, 8);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns(192, 168, 0, 1);

void setup() {
  Serial.begin(9600);
  connectWiFi();

  //tasks
  SoftTimer.add(&watchdogFeederTask);
  SoftTimer.add(&t1);
  SoftTimer.add(&t2);
}

void connectWiFi() {
  WiFi.config(ip, gateway, subnet, dns);
  WiFi.begin(ssid, wifiPass);

  Serial.println("WiFi status:");
  Serial.println(WiFi.status());

  int wifiStatus = WiFi.status();

  while (wifiStatus != WL_CONNECTED && wifiStatus != 6) {
    delay(500);
    Serial.print(".");
  }
}

void feedWatchdog(Task* me) {
  ESP.wdtFeed();
}

void callBack1(Task* me) {
  Serial.println("callBack1");
}

void callBack2(Task* me) {
  Serial.println("callBack2");
}
`

prampec, 2017-08-28:
I'm not sure right now how to handle this problem. As a workaround I recommend you the following change:
In the SoftTimer.cpp at line 96 there is a while(true) inifinite loop, that has a closing bracket on the 103. Please remove both lines! (So that the run() will exit, and relaunched by the loop() .)

prampec, 2017-08-28:
I have introduced an option for preventing problems like this. Please take a look at the updated documentation: https://github.com/prampec/arduino-softtimer/blob/wiki/WhyHow.md
Note that while this feature is available in the GIT repository, This option will be available in release 3.1.4 or higher.

tommyNy, 2017-08-28:
Ok, thx for your response. I commented a first while loop in SoftTimerClass::run() and it works!

TomaxMe, 2018-04-15:
I'm using WeMos D1 R2 mini with ESP 8266 module for controlling 8 Relays Module. I've deleted that 2 lines of code as you have been instructed above (in SoftTimer.cpp, line 96 and line 103). I can ping now succesfully.

The problem: 
I cannot see anymore the webpage (buttons) built in my Arduino's code after adding SoftTimer library.

This is my code:

 //https://www.youtube.com/c/SidsEClassroom

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <SoftTimer.h>

// -- Step 1 (SoftTimer.h): declare the function names gonna to be used.
void feedWatchdog(Task* me);
void callBack1(Task* me);
void callBack2(Task* me);

// Step 2 (SoftTimer.h): define tasks & time
Task watchdogFeederTask(200, feedWatchdog);
Task t1(3000, callBack1);
Task t2(5000, callBack2);

const char* ssid = "xxx";             //Set your wifi network name(ssid)
const char* password = "xxx";                //Set your router password
//IP is forwarded by OrangeRouter to: 192.168.1.103

// Declare pins & relay states

const int r1 = D1;
const int r2 = D2;
const int r3 = D3; 
const int r4 = D4;
const int r5 = D5;
const int r6 = D6;
const int r7 = D7;
const int r8 = D8;

String r1state = "Off";
String r2state = "Off";
String r3state = "Off";
String r4state = "Off";
String r5state = "Off";
String r6state = "Off";
String r7state = "Off";
String r8state = "Off";

WiFiServer server(80);
// Client variables 
char linebuf[80]; // by default: 80
int charcount=0;

void setup() {
  
  // Connect to serial port
  Serial.begin(9600); //change according to your com port baud rate  
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  // Connect to WiFi network
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  // Start the server
  server.begin();
  Serial.println("Server started.");

  // Print the IP address
  Serial.print("Use this URL : ");
  Serial.print("http://");
  Serial.print(WiFi.localIP());
  Serial.println("/");

  // Relay module prepared  
  pinMode(r1, OUTPUT);
  digitalWrite(r1, HIGH);
  pinMode(r2, OUTPUT);
  digitalWrite(r2, HIGH);
  pinMode(r3, OUTPUT);
  digitalWrite(r3, HIGH);
  pinMode(r4, OUTPUT);
  digitalWrite(r4, HIGH);
  pinMode(r5, OUTPUT);
  digitalWrite(r5, HIGH);
  pinMode(r6, OUTPUT);
  digitalWrite(r6, HIGH);
  pinMode(r7, OUTPUT);
  digitalWrite(r7, HIGH);
  pinMode(r8, OUTPUT);
  digitalWrite(r8, HIGH);

//Step 3 (SoftTimer.h): setup tasks
SoftTimer.add(&watchdogFeederTask);
SoftTimer.add(&t1);
SoftTimer.add(&t2);

}

//Step 4 (SoftTimer.h): run functions, replacing the Arduino's default loop().
// Display dashboard page with on/off button for relay

void dashboardPage(WiFiClient &client) {
  client.println("<!DOCTYPE HTML><html><head>");
  client.println("<meta name=\"viewport\" content=\"width=device-width, initial-scale=1\"></head><body>");                                                             
  client.println("<h2><a href=\"/\">SCHIT. SmartHome // TEST SOFTTIMER</a></h2>");

    
  // If relay == off, it shows the button to turn on   & viceversa    
  //Attention: keep the order if/off, if/off... for each relay, otherwise will not be able to control the buttons   
  if(r1state == "Off"){client.println("<a href=\"/relay1on\"><button>1-ON</button></a>");}
  if(r1state == "On"){client.println("<a href=\"/relay1off\"><button>1-OFF</button></a>");}

  if(r2state == "Off"){client.println("<a href=\"/relay2on\"><button>2-ON</button></a>");}
  if(r2state == "On"){client.println("<a href=\"/relay2off\"><button>2-OFF</button></a>");}

  if(r3state == "Off"){client.println("<a href=\"/relay3on\"><button>3-ON</button></a>");}
  if(r3state == "On"){client.println("<a href=\"/relay3off\"><button>3-OFF</button></a>");}

    if(r4state == "Off"){client.println("<a href=\"/relay4on\"><button>4-ON</button></a>");}
  if(r4state == "On"){client.println("<a href=\"/relay4off\"><button>4-OFF</button></a>");}

    if(r5state == "Off"){client.println("<a href=\"/relay5on\"><button>5-ON</button></a>");}
  if(r5state == "On"){client.println("<a href=\"/relay5off\"><button>5-OFF</button></a>");}

    if(r6state == "Off"){client.println("<a href=\"/relay6on\"><button>6-ON</button></a>");}
  if(r6state == "On"){client.println("<a href=\"/relay6off\"><button>6-OFF</button></a>");}

    if(r7state == "Off"){client.println("<a href=\"/relay7on\"><button>7-ON</button></a>");}
  if(r7state == "On"){client.println("<a href=\"/relay7off\"><button>7-OFF</button></a>");}

  if(r8state == "Off"){client.println("<a href=\"/relay8on\"><button>8-ON</button></a>");}
  if(r8state == "On"){client.println("<a href=\"/relay8off\"><button>8-OFF</button></a>");}

  // Generates buttons to control the relay
  client.println("<h4>Relay 1 = " + r1state + "</h4>");
  client.println("<h4>Relay 2 = " + r2state + "</h4>");
  client.println("<h4>Relay 3 = " + r3state + "</h4>");
  client.println("<h4>Relay 4 = " + r4state + "</h4>");
  client.println("<h4>Relay 5 = " + r5state + "</h4>");
  client.println("<h4>Relay 6 = " + r6state + "</h4>");
  client.println("<h4>Relay 7 = " + r7state + "</h4>");
  client.println("<h4>Relay 8 = " + r8state + "</h4>");
}

 void feedWatchdog(Task* me) {
ESP.wdtFeed();
}

void callBack1(Task* me) { 
Serial.println("callBack1");
  digitalWrite(r5, HIGH);

}

void callBack2(Task* me) {
Serial.println("callBack2");
  digitalWrite(r5, LOW);
}

prampec, 2018-04-16:
The theory is that you either let the "loop()" loop (what is disabled by default), or feed the watchdog regularly. You do not need both. On the other hand more feeding should not cause problem. So this is something to be investigate.

TomaxMe, 2018-04-16:
Looks pretty advanced to me. I just hope you'll figure out how to fix your work... which has a good potential.

prampec, 2018-04-16:
In short:
If you use WifiServer, you need to call the `server.handleClient();` method to maintain client request. So put this into the `void feedWatchdog(Task* me)` method. Also, you can set the period in the `watchdogFeederTask` down to 0 (zero).

prampec, 2018-04-24:
Release 3.1.4 was issued. Please check SoftTimer5DelayRunEsp8266WebServer example for using ESP8266.

closed 2018-04-24 by prampec