prampec/arduino-softtimer #23: Enhancement Request: Add asynchronous AnalogScanner functionality

bdutta, 2017-08-27:
Arduino-Softtimer is a great and super useful library, that I have used in the past. 

Lately, I am working on a project where I'd like to scan 12 analog channels on a Leonardo, apart from scanning digital pins (momentary push-buttons read with debounce), manage state of few LEDs, and push data over USB CDC to host.

There is this [AnalogScanner](https://github.com/merose/AnalogScanner) library that lets one use an asynchronous approach to dealing with several analog channels, which might go very well with this library. Would it be possible to merge or build similar capability to AnalogScanner ?

prampec, 2017-08-28:
Theoretically it is possible to build something similar to this with SoftTimer, as I did for the PCI interrupt handling. However currently I do not have the capacity to fulfill this request. Let's keep this issue open.

bdutta, 2017-08-28:
Is there a list of other Arduino libraries that SoftTimer is able to happily coexist with, in the same Sketch, or parameters on which to evaluate such other libraries ?

prampec, 2017-08-28:
No. Generally say SoftTimer is a relatively simply library. There are two key features that may deny to work with other libraries/platforms.
1. It does not launch the loop method the regular way. - I just have introduced a macro for overriding this option. See updated documentation: https://github.com/prampec/arduino-softtimer/blob/wiki/WhyHow.md
2. It depends on the PciManager library. - For this problem I have set up a side-repository with a version, that does not depends on the PciManager: https://github.com/prampec/arduino-softtimer-nopci The drawback of this is that you will not be able to access some powerful functions like the debouncer. (So for example you will not be able to use the debouncer provided by the SoftTimer library in combination with the SoftwareSerial, as both tries to implement the PCI listener methods.)

Overall, I do not see any point where SoftTimer interferes with other libraries, that cannot have a workaround. On the other hand, some libraries might use the delay() method, that goes against the philosophy of SoftTimer.