prampec/arduino-softtimer #24: ESP32 support or examples

debrucer, 2018-04-30:
Can you identify a reason for me that this library won't work on an ESP32? It was difficult to get the ESP32 boards to even show in the Arduino IDE, and now that they do, your useful library doesn't seem to work for it. Esp recommends that ESP32 devices be programmed in their environment, which must be why they make them so hard to access in the Arduino IDE. 

Can you enlighten me as to why I am getting my failure, please?

Everything was okay until...

C:\Users\debrucer\Documents\Arduino\libraries\SoftTimer\src\Debouncer.h:32:25: fatal error: PciListener.h: No such file or directory

compilation terminated.

Using library SoftTimer at version 3.1.3 in folder: C:\Users\debrucer\Documents\Arduino\libraries\SoftTimer 
exit status 1
Error compiling for board ESP32 Dev Module.

prampec, 2018-04-30:
Similar issues appeared when ESP8266 was introduced. I do own an ESP32, so I will check it.

worldspawn, 2019-07-20:
I resolved this by installing the library "PciManager".

stasyanok, 2019-12-21:
Same problem, pls check

stasyanok, 2019-12-21:
> Similar issues appeared when ESP8266 was introduced. I do own an ESP32, so I will check it.

The same problem, pls check

noamyogev84, 2023-08-06:
does not compile on Arduino MKR Zero

In file included from C:\Users\nyogev\AppData\Local\arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/Arduino.h:51:0,
                 from C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\SoftTimer.h:30,
                 from C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\FrequencyTask.cpp:27:
C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\FrequencyTask.cpp: In constructor 'FrequencyTask::FrequencyTask(int, float)':
C:\Users\nyogev\AppData\Local\arduino15\packages\arduino\hardware\samd\1.8.13\variants\mkrzero/variant.h:52:51: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
 #define portOutputRegister(port) (&(port->OUT.reg))
                                                   ^
C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\FrequencyTask.cpp:35:19: note: in expansion of macro 'portOutputRegister'
   _portRegister = portOutputRegister(digitalPinToPort(outPin));
                   ^~~~~~~~~~~~~~~~~~
In file included from C:\Users\nyogev\AppData\Local\arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/Arduino.h:51:0,
                 from C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\BlinkTask.h:30,
                 from C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\BlinkTask.cpp:27:
C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\BlinkTask.cpp: In member function 'virtual void BlinkTask::init()':
C:\Users\nyogev\AppData\Local\arduino15\packages\arduino\hardware\samd\1.8.13\variants\mkrzero/variant.h:52:51: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
 #define portOutputRegister(port) (&(port->OUT.reg))
                                                   ^
C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\BlinkTask.cpp:46:25: note: in expansion of macro 'portOutputRegister'
   this->_portRegister = portOutputRegister(digitalPinToPort(this->_outPin));
                         ^~~~~~~~~~~~~~~~~~
In file included from C:\Users\nyogev\AppData\Local\arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/Arduino.h:51:0,
                 from C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\SoftTimer.h:30,
                 from C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\SoftPwmTask.cpp:27:
C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\SoftPwmTask.cpp: In constructor 'SoftPwmTask::SoftPwmTask(int)':
C:\Users\nyogev\AppData\Local\arduino15\packages\arduino\hardware\samd\1.8.13\variants\mkrzero/variant.h:52:51: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
 #define portOutputRegister(port) (&(port->OUT.reg))
                                                   ^
C:\Users\nyogev\OneDrive - Intel Corporation\Documents\Arduino\libraries\SoftTimer\src\SoftPwmTask.cpp:39:19: note: in expansion of macro 'portOutputRegister'
   _portRegister = portOutputRegister(digitalPinToPort(pin));
                   ^~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1