prampec/arduino-softtimer #27: Compilation Error

eifarhi, 2019-07-10:
After installing the latest version of both softtimer and Pci maneger using Arduino's "Manage Libraries..." , I receive the following error message on compilation :
#####################################################################
Arduino: 1.8.8 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

libraries\SoftTimer\SoftTimer.cpp.o (symbol from plugin): In function `SoftTimerClass::remove(Task*)':

(.text+0x0): multiple definition of `loop'

sketch\Main_IO.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
#######################################################################

created from the simple following script:
########################################
#include <SoftTimer.h>
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
########################################

prampec, 2019-07-11:
Please try any of the examples! I also encourage you to read the documentation or the error message.

Error message: "_multiple definition of `loop'_"

Documentation: "_When you use SoftTimer you do not implement the "loop" function of the Arduino._"

https://github.com/prampec/arduino-softtimer/blob/wiki/QuickStartGuide.md

closed 2019-07-11 by prampec

virenchocha, 2022-07-03:
I got this error and tried to resolve but did not work. Here is what I have done 
1. Opened SoftTimer1Task.ino
2. Compiled and uploaded. LED started blinking as expected. Great !!
3. I noticed that there is no loop() in the .ino code. So I put one manually and there I got this error of duplicate loop()
4. Gone to documentation. Removed commented ENABLE_LOOP_ITERATION in SoftTime.h
5.  Commented SoftTimer.cpp loop() function
6. Complied and uploaded.
7. LEDs did not blink at all !

what is wrong i am dong

Here is the code (*.ino) with loop function mentioned in it.

`#include <SoftTimer.h>

// -- Define method signature.
void callBack1(Task* me);
void callBack2(Task* me);

Task t1(200, callBack1);
Task t2(100, callBack2);

void setup() {
  pinMode(13, OUTPUT);
  SoftTimer.add(&t1);
  SoftTimer.add(&t2);
}

void loop(){
  
}

void callBack1(Task* me) {
  digitalWrite(13, HIGH);
}
void callBack2(Task* me) {
  digitalWrite(13, LOW);
}`