prampec/arduino-softtimer #33: Regression of Rotary encoder

fredilarsen, 2020-09-29:
I have been using this library for some years, thanks.

I recently found that when reflashing a device that I have had in operation for years, the rotary encoder stopped working. After some trial and error I found that it works great with SoftTimer up to version 3.1.5 but not with 3.2.0 which is the last available in Arduino Studio.

Symptoms with v3.2: Rotary encoder movement not being picked up at all, at least the callback is not called.

prampec, 2020-09-29:
Have you noticed, that init() is now required to be called in setup? (See examples!)

fredilarsen, 2020-09-30:
Init was not called in my code, so I will add that and retest. Thank you.
I assumed the API was unchanged as it was only a minor version upgrade from 3.1.5 to 3.2.0.

prampec, 2020-09-30:
Well, yes. You might be right on that it was not emphasized well.
Thanks for the feedback any way!

closed 2020-09-30 by prampec

fredilarsen, 2020-10-01:
Thanks for the help!