prampec/arduino-softtimer #34: SoftTimer - TinyZero and SAMD21 Support

bibletech55, 2022-03-01:
Hi there,

I'm currently developing a project for which the Protothreading SoftTimer uses is ideal. The simplicity of SoftTimer is highly important in this code's design process to me. However, after setting up the code SoftTimer requires, I receive the error at the end of this post. I'm using a TinyZero board, a highly miniaturized version of the Arduino Zero (this is the link to the TinyZero's product site - https://tinycircuits.com/products/tinyzero-processor?variant=21137366515796). As part of my testing process before reporting, I tested both my project code and the included SoftTimer1Task example code - I believe both reported the same error. I have two questions:

Is SoftTimer compatible with SAMD21-based processors? If not, what would be a good SAMD21-compatible alternative library?

Is there anything I can do on my end to rectify these errors?

Thank you in advance for any help you can provide!

ERROR LOG:
-----------------------------------------------------------------
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "TinyZero, Default"

In file included from C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\cores\arduino/delay.h:23:0,

                 from C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\cores\arduino/Arduino.h:81,

                 from C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\BlinkTask.h:30,

                 from C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\BlinkTask.cpp:27:

C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\BlinkTask.cpp: In member function 'virtual void BlinkTask::init()':

C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\variants\tinyzero/variant.h:65:55: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

 #define portOutputRegister(port)   ( &(port->OUT.reg) )

                                                       ^

C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\BlinkTask.cpp:46:25: note: in expansion of macro 'portOutputRegister'

   this->_portRegister = portOutputRegister(digitalPinToPort(this->_outPin));

                         ^~~~~~~~~~~~~~~~~~

In file included from C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\cores\arduino/delay.h:23:0,

                 from C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\cores\arduino/Arduino.h:81,

                 from C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\SoftTimer.h:30,

                 from C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\FrequencyTask.cpp:27:

C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\FrequencyTask.cpp: In constructor 'FrequencyTask::FrequencyTask(int, float)':

C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\variants\tinyzero/variant.h:65:55: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

 #define portOutputRegister(port)   ( &(port->OUT.reg) )

                                                       ^

C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\FrequencyTask.cpp:35:19: note: in expansion of macro 'portOutputRegister'

   _portRegister = portOutputRegister(digitalPinToPort(outPin));

                   ^~~~~~~~~~~~~~~~~~

exit status 1

Error compiling for board TinyZero.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

prampec, 2022-03-02:
No panic! SoftTimer should work on any processor. Here we are dealing with a 32bit microcontroller where the port registers are stored in an uint32_t type.
If you take a look at the [BlinkTask.h](https://github.com/prampec/arduino-softtimer/blob/master/src/BlinkTask.h) you can see, that there is already a macro trying to solve this problem for ESP8266.
Ideally this macro should handle all 32 bit microcontrollers, but apparently I don't know what define to reference here.

What you can try is to change ESP8266 to a define matches your processor type. Again, not sure what define to use for SAMD21. Maybe SAMD21? :) (Please come back with the right phrase!)

As a worst case scenario you can delete all custom tasks prepared by SoftTimer, and start to create yours from scratch. Using only SoftTimer.h, SoftTimer.cpp, Task.h, Task.cpp.

bibletech55, 2022-03-03:
> No panic! SoftTimer should work on any processor. Here we are dealing with a 32bit microcontroller where the port registers are stored in an uint32_t type. If you take a look at the [...]

Thank you so much for your quick reply!

Just did some quick tests with the define ESP8266-macro that handles the registers - ARDUINO_SAMD_ZERO allows BlinkTask.h to work on the TinyZero! However, SoftTimer.h, FrequencyTask.cpp and SoftPWMTask.cpp still provide the same errors. 

I forgot to mention before that I didn't use BlinkTask.h in my custom code or the initial example I tested with. Perhaps SoftTimer.h requires the same macro? Would all the other headers and .cpps require the same macro and definition?

(Also, there are a lot of SAMD-based definitions - I found this definition and a lot of others in the SAMDTimerInterrupt library - not included in SoftTimer - the more definitions are included, the more SAMD21-based boards will be supported. :) )

I've attached the error log that occurs when replacing ESP8266 with ARDUINO_SAMD_ZERO in BlinkTask.h. :)

ERROR LOG
---------------------------------------------------------------
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "TinyZero, Default"

In file included from C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\cores\arduino/delay.h:23:0,

                 from C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\cores\arduino/Arduino.h:81,

                 from C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\SoftTimer.h:30,

                 from C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\FrequencyTask.cpp:27:

C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\FrequencyTask.cpp: In constructor 'FrequencyTask::FrequencyTask(int, float)':

C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\variants\tinyzero/variant.h:65:55: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

 #define portOutputRegister(port)   ( &(port->OUT.reg) )

                                                       ^

C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\FrequencyTask.cpp:35:19: note: in expansion of macro 'portOutputRegister'

   _portRegister = portOutputRegister(digitalPinToPort(outPin));

                   ^~~~~~~~~~~~~~~~~~

In file included from C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\cores\arduino/delay.h:23:0,

                 from C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\cores\arduino/Arduino.h:81,

                 from C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\SoftTimer.h:30,

                 from C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\SoftPwmTask.cpp:27:

C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\SoftPwmTask.cpp: In constructor 'SoftPwmTask::SoftPwmTask(int)':

C:\Users\isaac\Documents\ArduinoData\packages\TinyCircuits\hardware\samd\1.1.0\variants\tinyzero/variant.h:65:55: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

 #define portOutputRegister(port)   ( &(port->OUT.reg) )

                                                       ^

C:\Users\isaac\Documents\Arduino\libraries\SoftTimer\src\SoftPwmTask.cpp:39:19: note: in expansion of macro 'portOutputRegister'

   _portRegister = portOutputRegister(digitalPinToPort(pin));

                   ^~~~~~~~~~~~~~~~~~

exit status 1

Error compiling for board TinyZero.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

prampec, 2022-03-03:
Yes. Exactly the same deal with SoftPwmTask.h and FrequencyTask.h.

prampec, 2022-03-03:
Wait! I have forgot this... we have a branch called [NoPci](https://github.com/prampec/arduino-softtimer/commits/NoPci). This is the starting point of projects, are using MCUs other than 8bit AVR-s.
So use that branch!

prampec, 2022-03-03:
I have updated the [README](https://github.com/prampec/arduino-softtimer/blob/master/README.md) accordingly.

closed 2022-03-03 by prampec